tomer-w/victron_mqtt #6: Need to reconfigure with almost every update

MartinLeist, 2025-07-18:
Hi,

with almost every update the entityIds get overwritten or every entity gets newly discovered. Changing some of them manually doesnt help and with every update I needed to redo every connection, every dashboard and the energy overview because all entities were lost and new ones were created. Is there any way to prevent the integration from rediscovering the entities with every update anew?

MartinLeist, 2025-07-19:
Hi @tomer-w , thanks for the quick answer. I agree with you that changing this early is worth it. The last update also didn't change any entityIds anymore. Keep up the great work. I just got my victron system and your integration came at the right time for me and it really made getting it to work very easy. So thanks for providing this integration.

tomer-w, 2025-07-19:
@MartinLeist , happy to hear it is working for you. I will be glad to get any relevant feedback / improvement ideas.