tomer-w/victron_mqtt #8: Configure root topic

rehpa, 2025-07-18:
I want to start out by saying thank you for making this integration available. I'm a little new to mqtt so I might be overlooking something here.  I'm looking for a way when you add a new hub in home assistant for this integration to set the root topic.  Adding the cerbo itself as a hub works great, but I would rather have the hub point to my other mqtt broker that has a bridge setup to the cerbo.  That bridge puts all cerbo topics under the victron/ topic for organization.  I believe that is what causes me to be unable to add a hub using that broker as the source.  Is there a way to set the topic root when adding a hub?

[image]

tomer-w, 2025-07-19:
Fixed in: https://github.com/tomer-w/ha-victron-mqtt/releases/tag/v2025.7.12.
Let me know if it is working as expected.

rehpa, 2025-07-20:
It works!  Thank you for taking the time to implement this @tomer-w

tomer-w, 2025-07-20:
It was all done by AI. I just supervised :).