tomer-w/victron_mqtt #15: Negative solar reported...

zs3y, 2025-08-12:
1st of all, thank you very much for this integration.
Now for the issue, I don't know whether this is a Victron issue or not, but I'm hoping it can be solved...

Using home assistant I track solar produced from the MPPT's using the Yield today value for the Energy dashboard, since the GX device does not give combined energy yield, only combined power, which is fine, this will allow one to see which panels/MPPT does how well, problem is:

[image]

tomer-w, 2025-08-12:
Can you track it live using mqtt-explorer? I wonder what values you will be see there. Not sure if this is problem with data arriving from Victron or something wrong with the integration.

zs3y, 2025-08-12:
Hi, I think I posted this in the wrong section...  tomer-w ha-victron-mqtt might have been more correct, either way, I did install the mqtt-explorer, but now not seeing any Victron messages... HELP!

FYI the integration gives me this:

[image]
and the 85A MPPT has the following:

[image]

and as far as I can see both MPPTs dropped/reset their daily yield at 22:30 last night to 0

[image]

anyway, sorry if this is in the wrong section, still not sure how to let mqtt-explorer show the data we're looking at...

tomer-w, 2025-08-14:
You should install [mqtt-explorer](https://mqtt-explorer.com/), connect your laptop to the same netwrok you Cerbo is on and connect the app to your Cerbo IP address, port 1833. You should immediately start seeing the topics and the values they hold.
Let me know what specific issue you face if this not working, I might be able to give more specific help.

closed 2025-08-15 by tomer-w