tomer-w/victron_mqtt #28: Consuming from bridged topics does not work

msperl, 2025-08-24:
I tried to consume from a mqtt server that is replicating the topics from my GX device (read only)
and at least view_metrics is not working!

It complains about:
```
victron_mqtt.hub.CannotConnectError: Timeout waiting for first full refresh
```

This is a bit surprising - should we not be reading only?

Why do I need this: because I want to test those W/+ topics, which update only at irregular intervals and which are unfortunately not retained...
So development with some of topics is mostly waiting up to 15 minutes...

Hence I wanted to create a Read Only mirror that would have the topics set with the retained flag.

Config of the bridge in mosquito is:
```
connection victronmqtt2homeassistant
address <gx-device-ip>:8883
remote_username <user>
remote_password <pass>
bridge_cafile /etc/mosquitto/conf.d/victron-venusOS.ca.pem
bridge_insecure true
topic N/# in 0
topic R/# in 0
```

(So only in is supported)

msperl, 2025-08-25:
thanks for that documentation about venus MQTT!

it actually aided my understanding of how things work behind the scenes!

msperl, 2025-08-25:
maybe one thing:
* I got an instance already running in ha, which should do the ping.
* so from a performance perspective:
  * should there not be a means to skip that ping when developing?
  * maybe just wait for a ping or 2 (or just an initial ping and then rely on others to do that ping)

I have to say that when running the UI plus test I saw that VRM portal was complaining about too much load on my GX device (ekrano with 4 CPU cores) so it was no longer Realtime...

tomer-w, 2025-08-28:
Each client will send a different echo string and wait for its own echo string to come back so one client should not consume or interfere with reading data by another client.
With regards to load issue. I'm not aware of any specific problem around that. is Ekrano lower quality HW than Cerbo GX?

tomer-w, 2025-09-06:
Closing for now as I dont think there is a specific issue to fix here. Feel free to create new one for specific issue.

closed 2025-09-06 by tomer-w